merge/skulls #224: T540p Port?

K0ntrast, 2022-04-01:
Are there any plans to port Skulls to the T540p seeing as the T440p is already supported? I can't imagine that there would be much difference between the two.

I have an unused T540p which I can use for testing.

digmorepaka, 2022-04-19:
There is minimal difference between T540p and T440p, porting the T540p should be a matter of configuring the devicetree correctly. Here's a beginning of a port that you can contribute to. https://review.coreboot.org/c/coreboot/+/38204

gellatofresh, 2022-05-17:
T540p is a great machine.
When I've some time I'll contribute what I can... this and/or W54x.
I've heard claims of working coreboots on W541. I have less interest in that machine typically, however that will change should I disable it's dGPU through this.

Biggest difference aside from the express card would be the 1620p display, vs T440p