merge/skulls #225: skulls 1.0.4: build.sh wants to clone into '/home/coreboot/cb_build'

handmeatowel, 2022-04-12:
```
user@server:~/Downloads/skulls-1.0.4/x230$ pwd
/home/user/Downloads/skulls-1.0.4/x230
```
```
user@server:~/Downloads/skulls-1.0.4/x230$ ./build.sh 
1) ./free-defconfig-74d2218cc7
2) ./nonfree-defconfig-74d2218cc7
3) Quit
Please select a configuration to use for building or start with the -c option to use a different one: 2
Cloning into '/home/coreboot/cb_build'...
fatal: unable to access 'https://github.com/coreboot/coreboot.git/': Could not resolve host: github.com
fatal: not a git repository (or any parent up to mount point /home/coreboot)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: not a git repository (or any parent up to mount point /home/coreboot)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
```
Due to the script calling this non-existent path, building fails for all skulls targets.

Does someone have some pointers for me what needs to be changed?

handmeatowel, 2022-04-12:
**1.0.2** is the last version where this issue doesn't exist yet.

merge, 2022-05-30:
your problem seems to be with internet connection to github.com. /home/coreboot is inside of a docker container so should always exist during building.

closed 2022-05-30 by handmeatowel