merge/skulls #226: How to verify Skulls installation

freethinkpad, 2022-04-18:
Hello!

I wanted a Thinkpad with coreboot, so I bought a X230 with Skulls already installed. However, I fear that it could have been tampered. Could I verify the integrity of the installation?

I looked it up and the only thing I found is this comment on Reddit:
https://www.reddit.com/r/libreboot/comments/9dnj7b/how_can_i_verify_that_libreboot_hasnt_been/e6ellpj

I don't know if I could use this command to read the flashed image:
`flashrom -p internal -r file`

And then compare its hash SHA1 with the one of [free-defconfig-74d2218cc7](https://github.com/merge/skulls/blob/master/x230/free-defconfig-74d2218cc7) or [nonfree-defconfig-74d2218cc7](https://github.com/merge/skulls/blob/master/x230/nonfree-defconfig-74d2218cc7)

Could you give me a hand please? Thank you!

gellatofresh, 2022-05-17:
This is late but should another be reading this...
You need to gather the autors PGP key, and add it.
After that, download what will come in a pair such as <file.tar.xz> and <file.tar.xz.asc>
In terminal, ,
`-gpg --verify <ffile.tar.xz.asc>`

Of course I will assume one knows to cd to the containing directory. You could also drag and drop that file onto terminal instead of writing in whats contained <>

gellatofresh, 2022-05-17:
Ideally you can verify the provided hashes, if they are given this way. Then with confidence verify the checksum

merge, 2022-05-27:
yes, in principal that should work - although I'd recommend just flashing a trusted (and possibly newer) version instead. With Skulls, that's the only way to ensure the integrity. For real integrity verification, look into running Heads.

firefoxlover, 2022-12-25:
Where can we find your PGP key @merge ?

merge, 2022-12-27:
> Where can we find your PGP key @merge ?

that's my key: http://pgp.mit.edu/pks/lookup?op=get&search=0x500398DF5AB387D3

stafwag, 2023-03-12:
@merge Is your GPG key changed? I seems to be updated to 5352AD9973658B23BA379F702F708E4A5064D6B
Is this correct?

merge, 2023-03-17:
> @merge Is your GPG key changed? I seems to be updated to 5352AD9973658B23BA379F702F708E4A5064D6B Is this correct?

this is my new signing subkey, yes. F2082B880F9E423934686E3F500398DF5AB387D3 is the "key" where my (changing because I let them expire) subkeys belong to.

stafwag, 2023-03-20:
@merge ok. Nice perhaps just add link in README.md  to your homepage or another place were you publish the public key?

I used to verify it on https://martin.bet/ but your site seems to be down :-)

Thanks for your great work on this project BTW!

suddenlyfleck, 2023-06-30:
Dear @merge 

first and foremost: Thanks for your work! skulls is great and you provide such a streamlined experience here - I love it.

Trouble is: verifying the build. 

Solution would be: Please document a working way how to obtain your correct PGP key! Pleeease.

Here's the journey of my troubles:

```
gpg --verify skulls-1.0.8.tar.xz.asc skulls-1.0.8.tar.xz
gpg: Signature made Tue 13 Jun 2023 04:39:33 PM CEST
gpg:                using RSA key 45352AD9973658B23BA379F702F708E4A5064D6B
gpg: Can't check signature: No public key
```

ok lets get the key

```
gpg --keyserver keyserver.ubuntu.com --search-keys 45352AD9973658B23BA379F702F708E4A5064D6B
gpg --search-keys 45352AD9973658B23BA379F702F708E4A5064D6B
```

both don't work and give back 

```
gpg: key "45352AD9973658B23BA379F702F708E4A5064D6B" not found on keyserver
gpg: keyserver search failed: Not found
```

But this is the only information I get when downloading a release of yours - the asc file.

Maybe I can find out the fingerprint from that and then use `gpg --receive-keys`? (but honestly, who knows that shit from the top of one's head?)

Anyway, I google how to get the fingerprint and [find](https://askubuntu.com/a/151008) something like
```
gpg -nq --import --import-options import-show --with-colons skulls-1.0.8.tar.xz.asc
gpg: no valid OpenPGP data found.
```
looked strange, did not work. Surprise? Maybe not. 

Lets google to find your key. I land on this very page where I'm writing you now.
I find your answer here: https://github.com/merge/skulls/issues/226#issuecomment-1366117958

Great! Your key! Finally! I click the link you post and find the fingerprint "0x500398df5ab387d3"
So now:

```
gpg --receive-keys 0x500398df5ab387d3
gpg: key 500398DF5AB387D3: public key "Martin Kepplinger <martink@posteo.de>" imported
gpg: Total number processed: 1
gpg:               imported: 1
```
Finally! :laughing: 
```
gpg --verify skulls-1.0.8.tar.xz.asc skulls-1.0.8.tar.xz
gpg: Signature made Tue 13 Jun 2023 04:39:33 PM CEST
gpg:                using RSA key 45352AD9973658B23BA379F702F708E4A5064D6B
gpg: Can't check signature: No public key
```
No :disappointed: 
```
gpg --search-keys 0x500398df5ab387d3
gpg: data source: https://162.213.33.9:443
(1)	Martin Kepplinger <martink@posteo.de>
	Martin Kepplinger <martin.kepplinger@puri.sm>
	Martin Kepplinger <martink@posteo.at>
	  4096 bit RSA key 500398DF5AB387D3, created: 2015-03-20
Keys 1-1 of 1 for "0x500398df5ab387d3".  Enter number(s), N)ext, or Q)uit > 1
[... 66 lines cut ...]
>
>    To locate the key of a user, by email address:
>
>        gpg --auto-key-locate keyserver --locate-keys user@example.net

So I do

```
b@fedo ~/t/skulls> gpg --auto-key-locate keyserver --locate-keys martink@posteo.de
gpg: key 500398DF5AB387D3: public key "Martin Kepplinger <martink@posteo.de>" imported
gpg: key C100D7B57F2A1E26: public key "Martin Kepplinger <martink.tor@posteo.de>" imported
gpg: Total number processed: 2
gpg:               imported: 2
gpg: error retrieving 'martink@posteo.de' via keyserver: No fingerprint
b@fedo ~/t/skulls> gpg --verify skulls-1.0.8.tar.xz.asc skulls-1.0.8.tar.xz
gpg: Signature made Tue 13 Jun 2023 04:39:33 PM CEST
gpg:                using RSA key 45352AD9973658B23BA379F702F708E4A5064D6B
gpg: Can't check signature: No public key
```
I have started to doubt my life long before this point, but now it's hitting me hard :joy: 

By the way - I have two computers that use your skulls. (Thanks again for your work).
YESTERDAY, on the other computer, I managed to do this. Right now I'm taking already 30 minutes to do that, and YESTERDAY it took me 30 minutes. Yet, I have not sucessfully verified anything but the failure of my life's choice to be a user of PGP (sorry Phil, you did great and important work at your time).

So back on track:

suddenlyfleck, 2023-06-30:
The final solution was

going to [keys.openpgp.org](https://keys.openpgp.org/)

And downloading from there the key 45352AD9973658B23BA379F702F708E4A5064D6B

```
b@fedo ~/t/skulls> gpg --import F2082B880F9E423934686E3F500398DF5AB387D3.asc
gpg: key 500398DF5AB387D3: "Martin Kepplinger <martink@posteo.de>" 1 new signature
gpg: key 500398DF5AB387D3: "Martin Kepplinger <martink@posteo.de>" 1 new subkey
gpg: Total number processed: 1
gpg:            new subkeys: 1
gpg:         new signatures: 1
b@fedo ~/t/skulls> gpg --verify skulls-1.0.8.tar.xz.asc skulls-1.0.8.tar.xz
gpg: Signature made Tue 13 Jun 2023 04:39:33 PM CEST
gpg:                using RSA key 45352AD9973658B23BA379F702F708E4A5064D6B
gpg: Good signature from "Martin Kepplinger <martink@posteo.de>" [unknown]
gpg:                 aka "Martin Kepplinger <martin.kepplinger@puri.sm>" [unknown]
gpg:                 aka "Martin Kepplinger <martink@posteo.at>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: F208 2B88 0F9E 4239 3468  6E3F 5003 98DF 5AB3 87D3
     Subkey fingerprint: 4535 2AD9 9736 58B2 3BA3  79F7 02F7 08E4 A506 4D6B
     
````

suddenlyfleck, 2023-06-30:
I took the liberty to

```
gpg --keyserver keyserver.ubuntu.com --send-keys 45352AD9973658B23BA379F702F708E4A5064D6B F2082B880F9E423934686E3F500398DF5AB387D3 7346FE0227C33958813AC351C100D7B57F2A1E26
gpg: sending key 500398DF5AB387D3 to hkp://keyserver.ubuntu.com
gpg: sending key 500398DF5AB387D3 to hkp://keyserver.ubuntu.com
gpg: sending key C100D7B57F2A1E26 to hkp://keyserver.ubuntu.com
```

merge, 2023-06-30:
sorry, looks like the signing-subkey I used had not been pushed to the keyservers. thanks for verifying!

stafwag, 2023-07-01:
@merge Perhaps you can to documentation were/how to find/get public key to verify the download.
After this is done we can close this issue I guess :-)