merge/skulls #231: Garbled green artifacts on screen when selecting payload with non-free version

Gusher123, 2022-06-04:
After upgrading from version 1.14 to version 1.16, using the non_free "x230_coreboot_seabios_xxxxxxxxxx_top.rom" on my x230, selecting one of the two payloads (nvramcui and coreinfo) results in a garbled green output on the screen:

[image]
[image]

darnrain, 2022-06-05:
I had the same issue with Skulls v1.05. Just downgrade to Skulls v1.04 for now. I think it's a coreboot issue.

rxt1077, 2022-11-14:
I have the same issue.

XRevan86, 2022-12-12:
That's caused by ```CONFIG_YABEL_DIRECTHW=y``` being set in https://github.com/merge/skulls/commit/75792534343affd554cfda778996cdb2c8866232, not an upstream issue.

leftbit, 2022-12-15:
Had the same issue with Skulls v1.05 and v1.06. 
v1.04 still works for me.

pedrohqb, 2022-12-21:
I also confirm that. Turning to CONFIG_YABEL_DIRECTHW=n solves the issue? Maybe you should commit it, @XRevan86 .

se7en-x230, 2023-03-09:
I have the same problem 
I guess I have to build with the CONFIG_YABEL_DIRECTHW=n option?

KornelJahn, 2023-03-09:
I can confirm this problem, too.

Switched temporarily to the SeaVGABIOS free payload which solves the garbled green artifacts issue, however, it also forces some non-native 4:3 (?) resolution, making the bootsplash look quite squeezed horizontally.

commit e8c6edd5 (2024-02-28): x230: x230t: disable YABEL_DIRECTHW

Thank you https://github.com/Gusher123 who did a pull request, see
https://github.com/merge/skulls/pull/256

Closes: #231
closed 2024-02-28 by merge