merge/skulls #232: Feature request: Any hopes for the w530?

RHL120, 2022-06-05:
Hi, I have ordered a W530 but then found out that Skulls has not been ported to it. Are there any plans to port Skulls to the W530?
Thanks

darnrain, 2022-06-05:
You can use [](https://github.com/n4ru/1vyrain) to install custom stock bios on the w530.

RHL120, 2022-06-06:
> You can use https://github.com/n4ru/1vyrain to install custom stock bios on the w530.

Yes, I know but I want to flash coreboot.

student-sx, 2023-01-13:
If the T530 works, why not also the W530? What's the essential difference here?

RHL120, 2023-01-14:
> If the T530 works, why not also the W530? What's the essential difference here?

I have no idea, I am not bios expert

merge, 2023-02-06:
I'll try to add the W530 soon

RHL120, 2023-02-06:
> I'll try to add the W530 soon

Ok thanks

commit e777d8bf (2023-02-16): add the w530 the toplevel scripts

Fixes #232
closed 2023-02-16 by merge