merge/skulls #234: Background image appearing twice on boot - X230

boraboragoldfish, 2022-06-11:
Hi Martin. I have a freshly flashed X230 running the latest Skulls 1.0.5 and the background image appears twice during the boot. Is this something that I should be concerned? If so, should I re-flash the top image internally or externally to fix this issue? Thanks.

darnrain, 2022-06-13:
Same thing happen to me with skulls v1.05. I downgraded back to v1.04 tell the issue get's fixed.

compicat, 2022-10-01:
> Hi Martin. I have a freshly flashed X230 running the latest Skulls 1.0.5 and the background image appears twice during the boot. Is this something that I should be concerned? If so, should I re-flash the top image internally or externally to fix this issue? Thanks.

It is happening here too. The weird thing is that, even if you compile coreboot manually from the source, there is one blinking black screen before seabios load. It wasn't happening with the release in skulls v1.0.4, seabios was directly loading.