merge/skulls #237: T530--Error in instructions and issues recognizing 8MB chip

crispyambulance, 2022-06-30:
The first issue is that the pin layout for the T530 instructions to use the SOIC clip are backwards. Assuming the edge side is closer to you and the screen side away, the black and white diagram should be rotated 180 degrees. You can verify this by looking closely where the indented dot is in this photo from coreboot.org, which is also how my unit is. The indented dot is in the top left, not the bottom right like an X230. After I rotated my clip 180 degrees I could recognize the chips on my Pi.

https://www.coreboot.org/images/5/5d/T530_SPI_chips.jpg

I can also possibly get a better picture of this on my board.

--

The second issue I've run into now trying to flash two T530s is that the 8MB chip is not being read correctly. It is recognized as a 4MB chip, and each time flashrom lists it as a different chip when I run sudo flashrom -p linux_spi:dev=/dev/spidev0.0,spispeed=512 with outputs like this and others:

```
flashrom  on Linux 5.10.103-v7l+ (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Macronix flash chip "MX25L3205D/MX25L3208D" (4096 kB, SPI) on linux_spi.
No operations were specified.
```

```
flashrom  on Linux 5.10.103-v7l+ (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Macronix flash chip "MX25L3205(A)" (4096 kB, SPI) on linux_spi.
Found Macronix flash chip "MX25L3273E" (4096 kB, SPI) on linux_spi.
Multiple flash chip definitions match the detected chip(s): "MX25L3205(A)", "MX25L3273E"
Please specify which chip definition to use with the -c <chipname> option.
```

When I ran commands like this to write multiple copies of the chip

`sudo flashrom -p linux_spi:dev=/dev/spidev0.0,spispeed=512 -c MX25L3273E -r flash02.bin
`

I got 4 copies of the image on the chip and none of them were the same as any other when examined with md5sum.

Anyone have any idea what could be going on with this? Has anyone successfully flashed Skulls to the T530?

engineer22, 2023-06-29:
That's concerning. Came here to download for my T530 and decided to browse the issues only to come across this.

Did you figure it out, because maybe I should not start with this.

crispyambulance, 2023-06-29:
[image]

You need to bridge 2 pins on the 4MB chip while reading and writing to the 8MB chip. I did this by buying the devices in the picture I'm attaching, which I don't remember exactly what they're called, but the ebay listing was named

10x Multimeter SMD IC SOIC Mini Test Clip Hook Grabbers Probe Jumper 5 color XE

I connected the two "grabbers" using the same kind of cable as wiring the SOIC clip to my Pi, except male to male, not female to female. Sorry I'm not more literate on the names of stuff. I'm a bit of an amateur. It was finnicky and took a few tries to make the bridge functional because my cable didn't plug in neatly to the grabbers, but I got it to work eventually.

Once you can bridge those two pins, it's simply a matter of orienting the SOIC clip on the chip you're working on correctly. The directions for the T530 in their documentation show how it is for the X230. The T530 chips are rotated 180 degrees relative to that one. Just verify what you're doing by looking for the dot on the chip and you should be good. Even though I messed it up, it did not appear to damage the T530 that I made this mistake on repeatedly.

If you have any more questions I'll do my best to help.