merge/skulls #238: Strange behavior after updating to 1.0.5

cronos1998, 2022-07-01:
Hi,

I have updated Skulls on my Thinkpad X230.

The X230 came with the "non-free" version of Skulls 1.0.4 preinstalled (the version with the proprietary VGA BIOS) and now I flashed the free version of Skulls 1.0.5 (because I like the idea of free software).

Now there is some really strange behavior, though the device is booting the OS properly.

There's basically three strange things happening:

1) when starting the device, some messages (such as the version of SeaBIOS, the machine UUID as well as Debian's GRUB menu) are only shown on a part of the display.

It looks like the display is reduced to about a quarter of the actual size and it only changes to the full size after the GRUB menu has disappeared.

I noticed that this is partially mentioned in the README.md file and it advices people to put GRUB_GFXMODE=1366x768x32 into the /etc/default/grub file.

I have done that, but the GRUB menu is still shown in reduced size. Also, the messages about SeaBIOS and machine UUID right after startup are shown in reduced size.

2) the Skulls logo (the rabbit head) is shown with slightly wrong proportions. Width is reduced and height is increased.

3) after the GRUB menu has disappeared (when Debian or any other OS starts), there are the following messages appearing (this was not the case before updating Skulls):

[    2.710940]  i915 0000:00:02.0: [drm] *ERROR* uncleared fifo underrun on pipe A
[    2.710942]  i915 0000:00:02.0: [drm] *ERROR* CPU pipe A FIFO underrun
[    2.712118]  i915 0000:00:02.0: [drm] *ERROR* uncleared pch fifo underrun on pch transcoder A
[    2.712120]  i915 0000:00:02.0: [drm] *ERROR* PCH transcoder A FIFO underrun

Are these three things something to be concerned about or can they maybe be explained by the fact that I flashed the free version of Skulls?

I would like to stay with the free version a lot, unless there is really a need to go with the proprietary version.

Thanks in advance

darnrain, 2022-07-04:
Yeah just downgrade back to v1.04 for now, that's what I did.

henning-schild, 2022-07-05:
I guess you might want to compare non-free with free on a single release, so you do not end up mixing two changes. I have seen the "weird" graphics in early boot as well (both initially and in grub). So i am using non-free.

"free" probably always had those issues, and that has not much to do with 104 vs 105

cronos1998, 2022-07-06:
@darnrain
Are you using the free or the non-free version?

@henning-schild
So what do you think? Are the strange graphics just cosmetic issues or could they indicate some sort of critical malfunction?
Also, what do you think about the error messages regarding i915?  Are they related to the free version of Skulls as well?

cronos1998, 2022-07-06:
@merge
Can you please tell us your opinion?

henning-schild, 2022-07-06:
@cronos1998 i think it is cosmetic-only in the early phase for sure

for the error messages of i915, not sure what they are about and if the do any harm, when it comes to figuring out if they only appear in free but not in non-free ... why do you not go ahead and try? And if you find a correlation between them and "free" then open an issue just on that, maybe with multiple versions of skulls tried out.
Then someone can investigate what they mean and if they hurt, and would could be done. If they show always ... it is simply not related to skulls and should not be discussed here at all.

hypadr1v3, 2022-07-20:
Payloads do not work properly for me (shows weird tile glitches and I can't see anything apart from that) and Windows complains of ACPI_BIOS_ERROR with v1.0.5

compicat, 2022-08-31:
@hypadr1v3 same here

nuckle, 2022-09-03:
Had the same problem on x230. After installing `x230_coreboot_seabios` without `free` it got fixed

compicat, 2022-09-03:
> x230_coreboot_seabios

Did you make this via config file?

nuckle, 2022-09-07:
> > x230_coreboot_seabios
> 
> Did you make this via config file?
Ni, I did this via `sudo ./skulls.sh -b x230`. Just downloaded a skulls image with proprietary VGA bios

marwenius, 2023-03-21:
According to this, the ACPI-bug is fixed with Coreboot 4.18:
https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/message/VRAAQV5DRXKKFMBD5R7OC4BSZRM5HKCM/

With Skulls 1.0.7 I still get ACPI_BIOS_ERROR (ThinkPad X230).

I just tried it, rebuilt the images and adjusted the configuration file names (replaced db4b71ff10 with [f4c97ea131 / Coreboot 4.18](https://github.com/coreboot/coreboot/commit/f4c97ea131944c7be940b35361407e4b63a14faf) ... also tried [4d661eebe9 / Coreboot 4.19](https://github.com/coreboot/coreboot/commit/4d661eebe95b06acae7d99777ed36ba56a560112)).

Unfortunately, it still doesn't work with Windows in either case. It works with Fedora and Debian.

Didn't really expect it to just work like that. But I learned how to create the image myself, which is also nice :-)

Is there anything else I could try to fix the problem?

paulmenzel, 2023-03-22:
@marwenius, what ACPI error do you mean? I also do not see anything about Microsoft Windows in this Skulls issue. How is your comment related?

I recommend to create an issue in [coreboot’s issue tracker](https://ticket.coreboot.org/) or contact the coreboot mailing list, with an elaborate problem description and all the log files attached.

marwenius, 2023-03-22:
> @marwenius, what ACPI error do you mean? I also do not see anything about Microsoft Windows in this Skulls issue. How is your comment related?
>
> I recommend to create an issue in [coreboot’s issue [...]

I was referring to this comment:

> Payloads do not work properly for me (shows weird tile glitches and I can't see anything apart from that) and Windows complains of ACPI_BIOS_ERROR with v1.0.5

It is also reported there: https://github.com/merge/skulls/issues/230

Sorry if I used the wrong topic.