merge/skulls #239: Outstanding Performance from Skulls v1.04

darnrain, 2022-07-04:
I wanted to let everyone know that my x230 was running kinda slow and using a lot of battery power. I have a raspberry pi with a Pomona Clip, I removed Intel ME again.

git clone https://github.com/merge/skulls

hooked my clip up to the bottom flash

./external_install_bottom.sh -m d

then I flashed the rom skills v1.04 again on the top chip.

sudo ./skulls.sh -b x230 -i <rom image name>

Wow, my x230 is running so much faster and the battery draw is around 10.5w now as it was running about 15w to 20w normal. Not sure what happened. Maybe a bad flash the first time? Or maybe removing the Intel ME again did something. 

I followed this guide https://vimeo.com/177951809

He talks about not putting 3.3v into the chip, and just using WOL to power the chip for flashing. I guess its safer that way.

henning-schild, 2022-07-05:
Hey, can you please go a bit into detail. I also have an x230 and will update to 1.05 soon.

It should not really matter much if you flashed "internal" or with the clip, the outcome is the same and the clip only needed if internal flashing happens to be locked.

Which combination was slow, and which was fast? How did you measure the watts? Maybe there was something different going on ... like a background task going crazy for some other reason? Maybe you can repro and tell others how to reach that "slow, high watt" state.

I am not sure how exactly things work, but i think that when you are using skulls it should not matter at all what the content of that second chip is. It should simply not be needed/used because the BIOS will simply fit in the first chip.

henning-schild, 2022-07-05:
@darnrain i see you active in several issues, that is great! But also in other cases i had a hard time understanding what exactly you are trying to say. It is always a good idea to describe things in detail and cover things like "how to repro" "expected result vs actual result"

henning-schild, 2022-07-08:
Thanks for the details! 

I have skulls on an X230 and am currently under the impression that the content of the second chip does not matter at all, it will just be "dead code" because the stock ROM for the first chip was the only user. My custom kernel on gentoo likely does not have `mei*` modules, and if adding these modules would make the "dead chip" act again i would be surprised.

Say there is any kind of surprise call into ME only coming from the kernel, maybe documentation can be improved and it remains unclear whether it is a kernel issues or a skulls issue. From what i get from you i would say that one can either flush that second chip or not load `mei*` modules and skulls 1.04 will work "just fine", where 1.05 has issues not relevant here.

I will have to read docs and code (or even watch videos), maybe all that is already explained somewhere. And i might follow up and correct my current understanding.

But back to this issue. So you only see problems if the second chip has "stock" in it and the kernel uses `mei*`?

henning-schild, 2022-07-08:
Note that module blacklisting will have to be done in the initrd as well, to make really sure the offending modules are never loaded. Check `dmesg` and the contents of you initrd so the modules do not sneak in before your rootfs takes over.

darnrain, 2022-07-09:
> Thanks for the details!
>
> I have skulls on an X230 and am currently under the impression that the content of the second chip does not matter at all, it will just be "dead code" because the stock ROM [...]

Your welcome, 
Yep

darnrain, 2022-07-09:
I added this to grub and then updated.

GRUB_CMDLINE_LINUX_DEFAULT="quiet mei.blacklist=yes mei_me.blacklist=yes"

Is this correct?