merge/skulls #242: ./external_install_bottom.sh and 1vyrain possible?

marwenius, 2022-07-20:
I use 1vyrain on my X230 and have disabled ME via BIOS (Advanced Menu).

Can I now additionally use ./external_install_bottom.sh or can it cause problems if I don't use it in combination with ./external_install_top.sh?

Or in other words, is a mixture of 1vrain and skulls possible?

Thanks in advance!

marwenius, 2022-07-23:
I tested it and the good news in advance: the X230 continued to work. But: I could no longer get into the BIOS.

So I flashed back the original ROM and used me_cleaner without parameters: ME is gone, I can get into the BIOS and interestingly the option to disable ME in the 1vyrain BIOS has disappeared.

closed 2022-07-23 by marwenius

gellatofresh, 2022-07-25:
Really? This is possible without a programmer and flashing it physically?

tlaurion, 2022-07-25:
@gellatofresh No.
1vyrain is not a tool to flash skulls/Heads/coreboot.
1vyrain permits to deactivate ME while you are using 1vyrain. Once you decide to flash something else, you loose 1vyrain.

1vyrain is a tool taking advantage of vulnerabilities in original Lenovo bioses to bypass limitations, and permit to flash 4mb images (BIOS images). It will not unlock neither Intel Firmware Descriptor, nor touch ME region.

If you intend to use Skulls/Heads/Coreboot in the long run, you NEED an external pprogrammer. 
Skulls permits to unlock ifd through additional unlock parameter. You can clean ME by backuping ME and cleaning it prior of reflashing it back.

Heads prepare rom images that you can flash once externally to then flash internally. But no, again, your cannot user 1vyrain as a way to bypass external flashing of Skulls/coreboot/Heads if you also intend to use those bios images WITH ME disabled.

It's one, or the other, not both.