merge/skulls #246: ThinkPad X230 flashing using RPi does not work

felbinger, 2022-08-15:
I'm trying to flash the bottom chip of my X230, I double checked the wiring and it seams to be correct.
```
nicof2000@rpi:~/skulls-1.0.5$ sudo ./external_install_bottom.sh -f rpi -m -k ../BOTTOM_chip
Ok. Run this on a Rasperry Pi.
trying to detect the chip...
chip not detected.
flashrom v1.2 on Linux 5.15.32-v8+ (aarch64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
chip not detected. Please find it manually and rerun with the -c parameter.
```

I also tried to add the 3,3 Volts when all the other pins are connected, but no luck.

I never done this before, so any help would be appreciated...

felbinger, 2022-08-15:
[image]

CarsonAlberding, 2022-08-15:
The 1st thing I would try is to make sure that the wiring is correct especially because pin 1 on the chip (CS pin on pi) on the chip is on the bottom right.  Also make sure that the pins retract into the plastic when connecting the clip; there should be no pins sticking out. 

Another issue is that sometimes the chip has issues with the 3.3v not being exact.  If your stock lenovo bios still works, enable wake on LAN.  Then turn off the laptop and disconnect the battery but leave it plugged in.  Connect an ethernet cable from the pi to the ethernet port on the motherboard and the LEDs on the ethernet port on the motherboard should light up.  If they don't, then turn the laptop on and shut it down again and the LEDs should stay on.  Connect the clip without the 3.3v wire and try flashing again.

Myles24, 2022-08-15:
The 1st thing I would try is to make sure that the wiring is correct especially because pin 1 on the chip (CS pin on pi) on the chip is on the bottom right. Also make sure that the pins retract into the plastic when connecting the clip; there should be no pins sticking out.

Another issue is that sometimes the chip has issues with the 3.3v not being exact. If your stock lenovo bios still works, enable wake on LAN. Then turn off the laptop and disconnect the battery but leave it plugged in. Connect an ethernet cable from the pi to the ethernet port on the motherboard and the LEDs on the ethernet port on the motherboard should light up. If they don't, then turn the laptop on and shut it down again and the LEDs should stay on. Connect the clip without the 3.3v wire and try flashing again.

felbinger, 2022-08-17:
Thanks a lot, seams to work with power plugged in and wol enabled:
```
nicof2000@rpi:~/skulls-1.0.5$ sudo ./external_install_bottom.sh -f rpi -m -k ../BOTTOM_CHIP.bak
Ok. Run this on a Rasperry Pi.
trying to detect the chip...
Detected MX25L6406E/MX25L6408E.
make: Entering directory '/home/nicof2000/skulls-1.0.5/util/ifdtool'
make: Leaving directory '/home/nicof2000/skulls-1.0.5/util/ifdtool'
Intel ME will be cleaned.
The flash ROM will be unlocked.
Start reading 2 times. Please be patient...
flashrom v1.2 on Linux 5.15.32-v8+ (aarch64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) on linux_spi.
Reading flash...  
```
How long does this part usually take? It's been five minutes now, and nothing yet...

Myles24, 2022-08-17:
As long as the chip is detected, you should be good.  Typically, it takes pretty long for it to read the flash, especially because the wires are long; just wait it out. Shouldn't take longer than 10-15 min.

felbinger, 2022-08-17:
Alright, I keep you posted :)

felbinger, 2022-08-17:
Saw something from seabios, operating system is booting as always, looks good.

felbinger, 2022-08-17:
Thank you very much!

closed 2022-08-17 by felbinger