merge/skulls #253: Green Art-ifacts on Payloads

alexgill, 2022-12-15:
Hi. I noticed this after upgrading a couple of release back: 
When entering either 'nvramcui' or 'coreinfo', what's shown are green artifacts, the same _exact_ ones across two separate x230 laptops, so it's consistently problematic. Currently running the 1.0.6 release. I do notice a lot of battery drain when powered off, so maybe there's something related about things not being all right. Suggestions appreciated.

SCREENSHOTS:
[nvramcui](http://alexgill.com/IMG_20221214_180825.jpg)
[coreinfo](http://alexgill.com/IMG_20221214_180902.jpg)

XRevan86, 2022-12-15:
Duplicate of #231

alexgill, 2022-12-15:
Agreed, this is a dupclucate issue as pointed out. Thanks

closed 2022-12-15 by alexgill