merge/skulls #255: Version 1.06 still has COREBOOT_DMI_DATE "12/07/2022"

Gusher123, 2022-12-20:
Wrote version 1.06, rebooted, looked at dmidecode and was surprised to still see the 12/07/2022 date. Wrote 1.06 again and nothing changed. Checked .rom file and found COREBOOT_DMI_DATE set at "12/07/2022". Could not find COREBOOT_DMI_DATE in github

> $cat x230_coreboot_seabios_3c528f2830_top.rom | strings | grep 2022
> Wed Dec 07 04:52:10 UTC 2022
> #define COREBOOT_BUILD "Wed Dec 07 04:52:10 UTC 2022"
> #define COREBOOT_DMI_DATE "12/07/2022"
> #define ASL_VERSION 0x20220331
> Wed Dec 07 04:52:10 UTC 2022
> #define BUILD_TOOLS "gcc: (coreboot toolchain v2022-09-18_c8870b1334) 11.2.0 binutils: (GNU Binutils) 2.37"
> Wed Dec 07 04:52:10 UTC 2022
> $

Gusher123, 2022-12-22:
It is referenced by [smbios_write_type0()](https://doxygen.coreboot.org/d1/d70/arch_2x86_2smbios_8c_source.html#l00335) in Coreboot in [version.h](https://doxygen.coreboot.org/df/dfc/version_8h.html#a76ed236b7a392039f2dd91a3607f3393) and [version.c](https://doxygen.coreboot.org/d2/d0b/version_8c_source.html#l00037)

merge, 2025-01-27:
the toolchain is updated regularly and 2022 is not referenced anymore of course. In case you want to add something here, feel free to re-open. thank you!

closed 2025-01-27 by merge