merge/skulls #258: T440p internal flashing from 1.0.6 to 1.0.7 (Error: "Image size doesn't match the flash chip's size")

quincylinux, 2023-02-21:
A few weeks ago I externally flashed my T440p using version 1.0.6.

Today I tried to update internally to 1.0.7 with
`sudo ./skulls.sh -b t440p`
after rebooting with `iomem=relaxed`, but I got the following error:

`Error: Image size (12582912 B) doesn't match the flash chip's size (8388608 B)!`

I am not familiar with this error, but below are the full outputs of my flashing commands:

`sudo ./external_install_top.sh -b t440p -k <backup-file-to-create>`:

$ sudo ./external_install_top.sh -b t440p -k t440p_4m.bak
1) ./t440p_coreboot_seabios_3c528f2830_top.rom
2) Quit

Please select a file to flash or start with the -i option to use a different one: 1
Please select the hardware you use:
1) Raspberry Pi
2) CH341A
3) Tigard
4) Quit

Please select the hardware flasher: 2
trying to detect the chip...
Detected W25Q32.V.
verifying SPI connection by reading 2 times. please wait.
flashrom v1.2 on Linux 6.1.9-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25Q32.V" (4096 kB, SPI) on ch341a_spi.
Reading flash... done.
flashrom v1.2 on Linux 6.1.9-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25Q32.V" (4096 kB, SPI) on ch341a_spi.
Reading flash... done.
current image saved as t440p_4m.bak
connection ok. flashing t440p_coreboot_seabios_3c528f2830_top.rom
flashrom v1.2 on Linux 6.1.9-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25Q32.V" (4096 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.
DONE

`sudo ./external_install_bottom.sh -m -k <backup-file-to-create>`:

$ sudo ./external_install_bottom.sh -m -k t440p_8m.bak
Skulls

Please select the hardware you use:
1) Raspberry Pi
2) CH341A
3) Tigard
4) Exit

Please select the hardware flasher: 2
Ok. Connect a CH341A programmer
trying to detect the chip...
Detected W25Q64.V.
make: Entering directory '/home/quincy/Downloads/skulls-1.0.6/util/ifdtool'
gcc -O2 -g -Wall -Wextra -Wmissing-prototypes -Werror -I../commonlib/include -c -o ifdtool.o ifdtool.c
gcc -o ifdtool ifdtool.o 
make: Leaving directory '/home/quincy/Downloads/skulls-1.0.6/util/ifdtool'
Intel ME will be cleaned.
The flash ROM will be unlocked.
Start reading 2 times. Please be patient...
flashrom v1.2 on Linux 6.1.9-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25Q64.V" (8192 kB, SPI) on ch341a_spi.
Reading flash... done.
flashrom v1.2 on Linux 6.1.9-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25Q64.V" (8192 kB, SPI) on ch341a_spi.
Reading flash... done.
current image saved as t440p_8m.bak
connection ok
start unlocking ...
Full image detected
The ME/TXE region goes from 0x3000 to 0x500000
Found FPT header at 0x3010
Found 28 partition(s)
Found FTPR header: FTPR partition spans from 0x160000 to 0x210000
ME/TXE firmware version 9.1.45.3000
Public key match: Intel ME, firmware versions 9.0.x.x, 9.1.x.x
The AltMeDisable bit is NOT SET
Reading partitions list...
 PSVN (0x00000bc0 - 0x000000c00, 0x00000040 total bytes): removed
 FOVD (0x00000c00 - 0x000001000, 0x00000400 total bytes): removed
 MDES (0x00001000 - 0x000002000, 0x00001000 total bytes): removed
 FCRS (0x00002000 - 0x000003000, 0x00001000 total bytes): removed
 EFFS (0x00003000 - 0x0000df000, 0x000dc000 total bytes): removed
 BIAL (NVRAM partition, no data, 0x0000add3 total bytes): nothing to remove
 BIEL (NVRAM partition, no data, 0x00003522 total bytes): nothing to remove
 BIIS (NVRAM partition, no data, 0x00036000 total bytes): nothing to remove
 NVCL (NVRAM partition, no data, 0x000069c9 total bytes): nothing to remove
 NVCM (NVRAM partition, no data, 0x0000439b total bytes): nothing to remove
 NVCP (NVRAM partition, no data, 0x0000a3fe total bytes): nothing to remove
 NVHM (NVRAM partition, no data, 0x00000058 total bytes): nothing to remove
 NVJC (NVRAM partition, no data, 0x00003da0 total bytes): nothing to remove
 NVKR (NVRAM partition, no data, 0x00005fb4 total bytes): nothing to remove
 NVNF (NVRAM partition, no data, 0x0000175f total bytes): nothing to remove
 NVOS (NVRAM partition, no data, 0x0003a34b total bytes): nothing to remove
 NVSH (NVRAM partition, no data, 0x000022c0 total bytes): nothing to remove
 NVSM (NVRAM partition, no data, 0x00001de8 total bytes): nothing to remove
 NVTD (NVRAM partition, no data, 0x00001feb total bytes): nothing to remove
 NVUK (NVRAM partition, no data, 0x00008940 total bytes): nothing to remove
 PLDM (NVRAM partition, no data, 0x000043c5 total bytes): nothing to remove
 TMNN (NVRAM partition, no data, 0x000001a6 total bytes): nothing to remove
 GLUT (0x000df000 - 0x0000e8000, 0x00009000 total bytes): removed
 LOCL (0x000e8000 - 0x0000ec000, 0x00004000 total bytes): removed
 WCOD (0x000ec000 - 0x000160000, 0x00074000 total bytes): removed
 FTPR (0x00160000 - 0x000210000, 0x000b0000 total bytes): NOT removed
 NFTP (0x00210000 - 0x00048a000, 0x0027a000 total bytes): removed
 MDMV (0x0048a000 - 0x0004ca000, 0x00040000 total bytes): removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xda)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x1c6487 - 0x1c66b1       ): removed
 ROMP             (Huffman, fragmented data, \~1 KiB   ): NOT removed, essential
 BUP              (Huffman, fragmented data, \~70 KiB  ): NOT removed, essential
 KERNEL           (Huffman, fragmented data, \~226 KiB ): removed
 POLICY           (Huffman, fragmented data, \~99 KiB  ): removed
 ClsPriv          (LZMA   , 0x1c66b1 - 0x1c6a8a       ): removed
 SESSMGR          (LZMA   , 0x1c6a8a - 0x1d2413       ): removed
 SESSMGR_PRIV     (LZMA   , 0x1d2413 - 0x1d7d03       ): removed
 HOSTCOMM         (LZMA   , 0x1d7d03 - 0x1e0035       ): removed
 TDT              (LZMA   , 0x1e0035 - 0x1e53fa       ): removed
 FPF              (LZMA   , 0x1e53fa - 0x1e6ef9       ): removed
The ME minimum size should be 1560576 bytes (0x17d000 bytes)
The ME region can be reduced up to:
 00003000:0017ffff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Removing ME/TXE R/W access to the other flash regions...
Checking the FTPR RSA signature... VALID
Done! Good luck!
File /tmp/tmp.1jHLHWg0Dr/work.rom is 8388608 bytes
Writing new image to /tmp/tmp.1jHLHWg0Dr/work.rom.new
ifdtool and me_cleaner ok
make: Entering directory '/home/quincy/Downloads/skulls-1.0.6/util/ifdtool'
rm -f ifdtool *.o *~ .dependencies
make: Leaving directory '/home/quincy/Downloads/skulls-1.0.6/util/ifdtool'
start writing...
flashrom v1.2 on Linux 6.1.9-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25Q64.V" (8192 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.
DONE

`sudo ./skulls.sh -b t440p`:

$ sudo ./skulls.sh -b t440p
1) ./t440p_coreboot_seabios_db4b71ff10_top.rom
2) Quit

file not specified. Please select a file to flash. Please read the README for details about the differences: 1
t440p_coreboot_seabios_db4b71ff10_top.rom: OK
input: t440p_coreboot_seabios_db4b71ff10_top.rom
output: output/t440p_coreboot_seabios_db4b71ff10_top_prepared_12mb.rom
Warning: Make sure not to power off your computer or interrupt this process in any way!
         Interrupting this process may result in irreparable damage to your computer!
Flash the BIOS now? y/N: y
flashrom v1.2 on Linux 6.1.12-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using region: "bios".
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Opened /dev/mtd0 successfully
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Error: Image size (12582912 B) doesn't match the flash chip's size (8388608 B)!

merge, 2023-02-24:
I can't reproduce this. This is very strange. not sure whether Linux v6.1 does something differently in this case now - I'm still using v5.10 based kernels. Let's leave this issue open to check that.

quincylinux, 2023-03-07:
Ok so I installed Debian which uses Linux v5.10 and I was able to update internally to 1.0.7, so I think Linux v6.1 does something differently.

cthulux, 2023-03-29:
Hi! I have both x230 and t440p and had the same issue only in t440p (Archlinux  kernel 6.1.15-lts) which I was able to fix following [flashrom FAQ](https://www.flashrom.org/FAQ).

That is as root:

`modprobe -r lpc_ich`

Or oneline with update:

`modprobe -r lpc_ich; sleep 4; ./skulls.sh -b t440p`

No need to enable `iomem=relaxed` in grub parameters.

Hope it helps