merge/skulls #260: Boot process x230

se7en-x230, 2023-03-10:
I successfully flashed the latest version and have several question/problems

skulls logo appears for a split second then the SeaBios info and after that the skulls logo again.
Question: is that what supposed to happen?
                   If yes is there a way to suppress the SeaBios info screen?
 Problem: Both of the default payloads show only green boxes and I have to restart.

merge, 2023-03-17:
well. first coreboot shows the logo, then after the "seabios info", seabios shows it too :) that's currently how it is.

if the payloads don'T work that's strange and should actually be a coreboot bug, but I tested them here and they worked. Do you use the nonfree VBIOS ? thanks

se7en-x230, 2023-03-17:
seabios_free_xyz_top.rom works
seabios_xyz_top.rom give the artifacts

pedrohqb, 2023-03-18:
It has been while we have this bug. We already have an working commit for that: https://github.com/merge/skulls/pull/256