merge/skulls #266: Please use flashrom v1.2 or later. You seem to use .

SKELETONTOPROM, 2023-04-06:
My flashrom version is:

flashrom is already the newest version (1.3.0-2.1)

But I get this error when trying to update:

sudo ./skulls.sh -b x230
[sudo] password for myusername: 
Please use flashrom v1.2 or later. You seem to use .

matsura8, 2023-06-02:
I had the same problem (In my case, Kali 2023. When I ran 'flashrom -v' the output declared it was an unspecified version number even though it stated v1.3 when installing with apt. The version number must have not been specified properly in the release. I would try installing from another source or compiling your own flashrom files. Flashrom also has precompiled binaries for most distros.

matsura8, 2023-06-04:
It seems like ~/flashrom-1.3.0/util/getversion.sh --version is returning undefined all the time, so it would be ideal to modify code keep this issue from persisting in all other builds.

matsura8, 2023-06-04:
The updated code though still gives error and does not allow to proceed. It appears the lack of version number was included in their release tarball. I have notified flashrom dev already.

matsura8, 2023-06-04:
After repairing the flashrom code myself...
[image]

commit 60327b73 (2023-06-04): functions: fix flashrom version check

flashrom dropped the "v" prefix for version strings.

Fixes: #266
commit c9d38ab4 (2023-06-13): util: remove flashrom version check

First of all, the check is not supposed to be hidden in the
"check for root" function.

Secondly, flashrom changes how it displays version information and
this is simply wrong.

Fixes #266
closed 2023-06-13 by merge