merge/skulls #274: Request: Support for the ThinkPad T430s

K0ntrast, 2023-10-09:
I have recently acquired a ThinkPad T430**s** laptop that I want to Coreboot. I am of the understanding that it is supported by Coreboot, but the [T430](https://www.coreboot.org/Board:lenovo/t430s) and [T430s](https://doc.coreboot.org/mainboard/lenovo/t430.html) are two different branches. Would @merge be interested in supporting the T430**s** with Skulls 1.0.10?

I am happy to help and very willing to test it. Can send you a ROM dump to get started.

K0ntrast, 2023-10-09:
Also, should the [read me for the T430 Skulls installation](https://github.com/merge/skulls/tree/master/t430) specify that it is/isn't suitable for the T430**s**?

merge, 2024-02-20:
the same goes for similar questions: if I'd do it, I'd need such a laptop myself. Otherwise you'd have to port Skulls, test it and work in a pull request from there - if you're willing to test future releases too - i.e. if it's not a "production" laptop.