merge/skulls #275: "Error: Extra parameter found" when Flashing Bottom Chip T430

mansjals, 2023-10-20:
Hello!
I flashed top chip successfully.
I get this message when trying to flash the bottom chip:

xx@raspberrypi:~/skulls-1.0.9 $ sudo ./external_install_bottom.sh -m -k t430-bottom.bak
Skulls

Please select the hardware you use:
1) Raspberry Pi
2) CH341A
3) Tigard
4) Exit
Please select the hardware flasher: 1
Ok. Run this on a Rasperry Pi.
trying to detect the chip...
Detected W25Q64BV/W25Q64CV/W25Q64FV
W25Q64JV-.Q.
make: Entering directory '/home/mj/skulls-1.0.9/util/ifdtool'
make: Leaving directory '/home/mj/skulls-1.0.9/util/ifdtool'
Intel ME will be cleaned.
The flash ROM will be unlocked.
Start reading 2 times. Please be patient...
flashrom v1.3.0 on Linux 6.1.0-rpi4-rpi-v8 (aarch64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Error: Extra parameter found.
Please run "flashrom --help" for usage info.

How can I get around this error message?
Thank you for your help.

mansjals, 2023-10-21:
the top chip (for BIOS) was detected as MX25L3206E/MX25L3208E.

CrashAndSideburns, 2023-10-28:
I'm having an identical issue. Possibly related/similar to #52?

CrashAndSideburns, 2023-10-28:
Quick update: I used a swab and some alcohol to clean off the IC and was just barely able to read the lettering on it. My laptop's bottom chip is a W25Q64CV, so I just went into the script and manually overwrote the chip type. This is obviously not a proper solution, but it may be workable until someone more knowledgeable than me can fix this.

mansjals, 2023-10-28:
Thank you for your comments. And did you successfully clean ME from the bottom chip after you manually overwrote the chip type?

CrashAndSideburns, 2023-10-29:
Flashing the bottom chip succeeded after I modified the script. I don't recall the exact modification, but it was just something like adding the line `CHIPNAME="W25Q64CV"` (I don't know if your chip is necessarily the same one).

mansjals, 2023-10-29:
I read through the script "external_install_bottom.sh". I noticed that with the -c option you can specify a chip in the command line already. I could not find any inscription on the chip (or it was completely unreadable). Therefore I tried to run the script with -c option with all chips (W25Q64BV/W25Q64CV/W25Q64FV/W25Q64JV-.Q), one chip after another. The last chip W25Q64JV-.Q worked.
My machine is now running on coreboot.
I really appreciate your help.

CrashAndSideburns, 2023-10-30:
@mansjals you should probably reopen this issue. Even if we were able to get our respective cases sorted out, it is something that probably needs to be fixed in the scripts for others.

JM1705, 2023-11-16:
I got this same error, and resolved it by entering "W25Q64BV/W25Q64CV/W25Q64FV" as an option with -c

merge, 2023-11-23:
thanks for reporting. there are definitely chips missing in the script. That can basically only be done iteratively on an empirical basis :)  so @mansjals you had `W25Q64JV-.Q` working? @CrashAndSideburns had `W25Q64CV` Can you confirm the exact model name, and I'll add those 2 at least.

CrashAndSideburns, 2023-11-26:
@merge sorry for the slow reply! I can confirm that it was exactly `W25Q64CV`.

mansjals, 2023-12-25:
Yes, for me worked W25Q64JV-.Q.
Thank you for your great software!

native-son, 2024-01-04:
For what it is worth I just got this same error from running `./external_install_bottom.sh` -m -k `<backup-filename>` on a T440p.

I also resolved it by re-running it as `./external_install_bottom.sh -c W25Q64BV/W25Q64CV/W25Q64FV -m -k <backup-filename>`

closed 2025-01-24 by merge