merge/skulls #279: T530 ACPI_BIOS_ERROR 1.0.9 WIN BOOT

yrvc38b8, 2023-11-24:
hello, I need help to fix this issue in the title. Has anyone an idea why windows doesn´t boot? I thought that coreboot fixed this issue. Flashed it successful and debian works great, but I can´t boot windows without this ACPI error. I also tried to flash 1.0.4 but I will get the same error.

xcpn, 2023-11-30:
Hey, I can't help with that but I have a question.

Did you manage to flash the T530 with no problems, only based on the documentation? I read [this issue](https://github.com/merge/skulls/issues/237#issuecomment-1613885787) and the workaround this user had to do to make it work, and wanted to know if you had this with your laptop.

Were the documentation steps provided enough for successful flash **on both chips**?

xcpn, 2023-12-01:
@merge please update the documentation for the T530 with this information.

xcpn, 2023-12-11:
@yrvc38b8 flashed a T430 and while Debian 12 works perfectly fine, Windows 10 is looping.

I didn't get any errors like you but mine is just looping. Something's going on with Windows.

yrvc38b8, 2023-12-17:
no I can´t fix it yet, I still need help

JannF, 2023-12-30:
Have the same problem. Windows 10 is looping with ACPI_BIOS_ERROR BSOD.

The two flashes were flawless with a tigard and a pomona clip.

merge, 2024-03-05:
> @merge please update the documentation for the T530 with this information.

we can do that, sure. I flashed T530 without this workaround until now though. Feel free to create a pull request.

hamensman, 2024-06-03:
Same issue with W530. Would be nice if they had builds with vbios.