merge/skulls #281: X230 with Quad-Core (Intel Core i7-3612QE) is slow

moerkey, 2024-02-01:
Hello,

I bought an X230 motherboard with quad-core (Intel Core i7-3612QE) from Asia and had Coreboot installed via the provider https://shop.vikings.net/product-category/coreboot-flashing-services/. Unfortunately, the system has now become much slower. Is it possible that the Coreboot is not set for an Intel Core i7-3612QE on an X230?

I have created some benchmarks. Here are the results with Phoronix.

Quad-Core CPU-Governor OnDemand: https://openbenchmarking.org/result/2311304-NE-KERNELDEM53
Quad-Core CPU-Governor Performance: https://openbenchmarking.org/result/2311303-NE-KERNELPER02
Dual-Core CPU-Governor Performance: https://openbenchmarking.org/result/2312226-NE-KERNELPER58

Here are also the tests via CLI under Linux with sysbench.

The tests ending with 1.txt were performed on the quad-core with conventional BIOS.
The tests ending with 2.txt were carried out on the quad-core with Coreboot & CPU-Governor OnDemand.
The tests ending with 3.txt were performed on the quad-core with Coreboot & CPU-Governor Performance.
The tests ending with 4.txt were carried out on the dual-core with conventional BIOS & CPU Governor Performance.

[cpu1.txt](https://github.com/merge/skulls/files/14128195/cpu1.txt)
[cpu2.txt](https://github.com/merge/skulls/files/14128196/cpu2.txt)
[cpu3.txt](https://github.com/merge/skulls/files/14128197/cpu3.txt)
[cpu4.txt](https://github.com/merge/skulls/files/14128198/cpu4.txt)
[io1.txt](https://github.com/merge/skulls/files/14128199/io1.txt)
[io2.txt](https://github.com/merge/skulls/files/14128201/io2.txt)
[io3.txt](https://github.com/merge/skulls/files/14128202/io3.txt)
[io4.txt](https://github.com/merge/skulls/files/14128203/io4.txt)
[ram1.txt](https://github.com/merge/skulls/files/14128205/ram1.txt)
[ram2.txt](https://github.com/merge/skulls/files/14128206/ram2.txt)
[ram3.txt](https://github.com/merge/skulls/files/14128207/ram3.txt)
[ram4.txt](https://github.com/merge/skulls/files/14128208/ram4.txt)

The RAM remained roughly the same speed on all systems.
The SSD remained the same speed on the quad-core with Coreboot, although the dual-core is a tick faster. This can be neglected.

The CPU has become drastically slower on the quad-core with Coreboot. The dual-core is a tick faster overall. However, this is due to the test and the CPU speed of 3.6 GHz compared to the quad-core.

The tests with sysbench can be carried out with the following commands.

sysbench --test=cpu run
sysbench --test=memory run
sysbench --test=fileio --file-test-mode=seqwr run

Thank you in advance.

Cheers

merge, 2024-05-16:
well that's a question for coreboot upstream, i.e. https://ticket.coreboot.org/projects/coreboot but it's very well possible that the x230 board code assumes that the vendor-sold platforms, https://thinkwiki.de/X230 are being used.

porterdcarnegie, 2024-11-27:
I did a few timed Linux 6.12 compilations on my Thinkpad x230, i7 3615QE. At first on a minimal linux distribution (KISS Linux) which resulted in 843 seconds with Coreboot and 847 seconds with original firmware (2.77). Notably, with Coreboot on Debian(testing), booted from a live USB, the kernel compiled in 505 seconds. 

On KISS Linux the reported cache size and cpu speed are as follows:
/sys/devices/...cpu0.../index3/size reports 6144K
/proc/cpuinfo reports 3092 MHz during load

I power the laptop from a power supply with a digital readout. It suggests 2-4W higher power consumption on Debian. Which kernel configuration option could be the culprit?

mara-ken, 2025-05-14:
I'm still interested and I can't really find anyone else talking about this.