merge/skulls #288: Error update skulls

Kzlztan, 2024-03-30:
[image]
[image]

how can I solve this problem?

merge, 2024-04-02:
try when being logged in as root, without using sudo. does the same error happen?

Kzlztan, 2024-04-02:
[image]
same error

Kzlztan, 2024-04-03:
sudo grub2-mkconfig -o /boot/grub2/grub.cfg
after I use this, apparently the update was successful.

how can I check i succeed to update?

Kzlztan, 2024-04-07:
thank you. it was new version

closed 2024-05-16 by merge