merge/skulls #289: X230 heat problem

Kzlztan, 2024-04-07:
I'm in trouble because I installed Skulls on the X230.
If I use Qubesos, it will soon exceed 100 ° C.
When I repaint the CPU grease, it became better, but it is still hot.
I think the cause is that the control of the CPU fan has deteriorated.
Is there a way to improve the control of CPU fans?

cikisir, 2024-04-16:
@Kzlztan I have the same problem with T530. I amateurishly limit the CPU frequency by using a command-line program called cpupower. It has init scripts for whichever init system you are using so that you don't have to write it yourself.

In the beginning, I thought I couldn't apply the thermal paste properly. It took a while for me to notice that coreboot is the main cause. Compiling programs with all CPU cores became dangerous since PC shuts itself off after a while. Limiting CPU frequency with cpupower works for compiling programs. But I sometimes remove the limiter and speed up the fans manually, in case where I am not going to stress the machine.

I hope this helps you. If you find a neater way to fix this problem, I really would like to hear the solution.

tlaurion, 2024-04-16:
Qubesos doesn't distribute coufeq kern module.
I would recommend trying existing solutions  https://www.thinkwiki.org/wiki/How_to_control_fan_speed but to be honest, having sold x230 for years, shipped with qubesos, seems like you fan is the cause of your issue.

leftbit, 2025-01-27:
Running on a x230 without heat problems - I did reapply thermal paste. So this might be hardware related.

merge, 2025-01-27:
I'd assume so as well. Let's not close this issue just yet but I've not seen any differences with overheating. AFAIK coreboot can't really make difference in this regard.