merge/skulls #290: Missing Virtualization and Graphic on T440p

NinaNoor, 2024-04-30:
I tried to flash the skulls on my thinkpad t440p by following the guide but it always fail for whatever reason and decided to flash using windows machine. I successfully backup both of the bios chip, flash the cleaned bottom chip with me _cleaner, and flash the top chip t440p_coreboot_seabios_04d6eb1eae_top.rom using windows machine too. it able to turned on, it able to boot debian live usb, in fact i wroting this issue on debian live usb but when i tried to install qubes os i got warning missing virtualization so i changed my mind to install debian, when i tried to install debian i got an error "Undefined video mode number: 314". how to solve both these error? i just wanted to install qubes os, did i do anything wrong? please help

merge, 2024-05-08:
It might be possible that you need to choose the normal Debian installer (not graphical), and select one resolution for the installation yourself. This is no issue after you've installed Debian Linux later.

I've never installed Qubes so I can't say, but I'd be very surprised if virtualization features would not work. At least as soon as Linux takes over.

NinaNoor, 2024-05-12:
Sucessfully using debian as my daily driver, i figure out my cpu is not supporting VT-d, thats why i got the error. I apologise for my ignorance and not checking my hardware before asking

closed 2024-05-12 by NinaNoor

merge, 2024-05-14:
np thanks for letting us know