merge/skulls #291: coreboot configuration utility & Skulls splash screen in eternal loop.

Backspin7792, 2024-05-21:
I successfully flashed a t440p with the latest version of skulls but when I restarted it the coreboot configuration utility appears, I press F1 to confirm and I'm at the skulls splash screen... once finished loading the utility starts again and so on again ..

The coreboot configuration utility settings are:

boot_option: Normal
reboot_counter: 0
debug_level: Debug
nmi: Disable
power_on_after_fail: Disable
first_battery: Primary
wlan: Enable
trackpoint: Enable
fn_ctrl_swap: Disable
sticky_fn: Disable
usb_always_on: Disable
backlight: Keyboard
f1_to_f12_as_primary: Enable
enable_dual_graphics: Disable
volume: 3

If i press ESC in the splash screen i get:

SeaBIOS (version rel-1.16.3.-0-ga6ed6b70)
Select boot device:

1. AHCI/0: P3-512 ATA-11 Hard-Disk (476 Gibytes)
2. Payload [nvramcui]
3. Payload [coreinfo]
t. TPM Configuration

 
There's i can do/change to fix the problem and load the OS ?

merge, 2024-05-21:
that simply sounds like on the "AHCI/0..." device is no bootable OS. Have you tried to install one?

Backspin7792, 2024-05-21:
i overwrited the old os and now it works.. maybe grub issue, thanks for the reply and for your works.

closed 2024-05-21 by Backspin7792