merge/skulls #308: T430 Error: Extra parameter found.

azide01, 2025-03-24:
This is happening on Debian 12 while attempting to flash my Thinkpad T430

sudo ./external_install_bottom.sh -m -k dude.bak
Skulls

Please select the hardware you use:
1) Raspberry Pi	 3) Tigard	  5) Exit
2) CH341A	 4) FT232H
Please select the hardware flasher: 2
Ok. Connect a CH341A programmer
trying to detect the chip...
Detected W25Q64BV/W25Q64CV/W25Q64FV
W25Q64JV-.Q.
make: Entering directory '/root/skulls/util/ifdtool'
gcc -O2 -g -Wall -Wextra -Wmissing-prototypes -Werror -I../commonlib/include -c -o ifdtool.o ifdtool.c
gcc -o ifdtool ifdtool.o 
make: Leaving directory '/root/skulls/util/ifdtool'
Intel ME will be cleaned.
The flash ROM will be unlocked.
Start reading 2 times. Please be patient...
flashrom unknown on Linux 6.1.0-32-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Error: Extra parameter found.
Please run "flashrom --help" for usage info.

azide01, 2025-03-24:
issue is very similar to https://github.com/merge/skulls/issues/52#issue-398683213

azide01, 2025-03-28:
issue can be solved using same solution as https://github.com/merge/skulls/issues/275#issue-1954956302

merge, 2025-04-02:
> issue can be solved using same solution as [#275 (comment)](https://github.com/merge/skulls/issues/275#issue-1954956302)

was it `W25Q64CV` for the T430? let's add that one to our script.