merge/skulls #313: T480-T480s Support

compicat, 2025-04-15:
Hi, hope you are all well. Would you consider adding T480 and T480s support?

merge, 2025-04-15:
thanks for the question. I personally, though, don't really have an incentive for doing that. On that platform, it's mostly about Intel's bootguard being enabled and that it can be disabled. There are plenty of other laptops with much newer Intel processor platforms that have bootguard disabled in the first place, so that's no reason for me.

More importantly yet, for the platforms newer than Haswell, a coreboot build always needs Intel's "firmware support package" that can be of significant size. I personally won't run a coreboot "shell" around nonfree firmware.

A nicely configured T430 is not much slower than a T480 as well. But again, this is me *personally*. I'm very happy to include support, if it is in coreboot upstream (or soon) and somebody else commits to maintain it for the future.

compicat, 2025-05-11:
@merge thanks for detailed and honest reply. Wish you best and thanks for amazing work again.

AndersonTorres, 2025-09-25:
Good night, people!
I found this by looking the Coreboot webpage.
If anything, recently I have bought a T480, and I wanna help.