merge/skulls #320: MSR locked

mmaalvarez, 2025-07-20:
I have flashed skulls (with SeaBIOS) in a Thinkpad T430 that runs Ubuntu 22.04, with a i7-3840QM CPU. I am trying to undervolt the CPU, but it seems that the msr is locked, as I can't write to it:

```
$ sudo wrmsr 0x150 0x0

wrmsr: CPU 0 cannot set MSR 0x00000150 to 0x0000000000000000

```
I checked that msr is loaded

```
$ lsmod | grep msr

intel_rapl_msr         20480  0

intel_rapl_common      40960  1 intel_rapl_msr

msr                    12288  0
```

and that I have sudo permissions

```
$ ll /dev/cpu/*

/dev/cpu/0:

total 0

drwxr-xr-x  2 root root     60 jul  9 11:31 ./

drwxr-xr-x 10 root root    200 jul  9 09:06 ../

crw-------  1 root root 202, 0 jul  9 09:06 msr

`
... (same in all cores)

and the output of /sys/kernel/security/lockdown is

```
$ cat /sys/kernel/security/lockdown

[none] integrity confidentiality
```

which afaik the [none] within brackets means that it is not locked?

merge, 2025-07-22:
I don't have experience with this, first of all, I'd do this as the root user, not using sudo. Secondly, I'd check how this is mapped and whether you need something like `iomem=relaxed` on the kernel cmdline

mmaalvarez, 2025-07-22:
Thanks for your suggestions, I understand that my issue is quite niche.
I do have iomem=relaxed in /etc/default/grub
`GRUB_CMDLINE_LINUX_DEFAULT="iomem=relaxed `
and running wrmsr as root user doesn't seem to make a difference, unfortunately.
`$ sudo su
root@user-ThinkPad-T430:/home/user# wrmsr 0x150 0x0
wrmsr: CPU 0 cannot set MSR 0x00000150 to 0x0000000000000000`